SICKAG/sick_scan_xd #476: LiDAR sensors: multiScan100 MULS1AA-112211 multiScan136 IP

Nervxz, 2025-04-11:
Hello, I want to know that is there any way to config hardly the IP of LiDAR ? because when I power off LiDAR and turn it back on it will require me to config the IP from SOPAS engineering tool on window. Mine is dual boot so whenever the LiDAR come up, I have to change to Window and config then turn back to linux.

rostest, 2025-04-15:
Hi,

Yes, it's possible to configure the LiDAR's IP address so that it persists across power cycles. From your description, it sounds like the IP is currently being set temporarily (in volatile memory), which is why it resets after every reboot.

To make the IP setting permanent:

Open SOPAS Engineering Tool on Windows.

Connect to the LiDAR and set the desired IP address as you normally do.

After setting the IP, make sure to click "Store to Device" (or similar wording, depending on your device model).
This step writes the configuration to the device’s non-volatile memory.

Power cycle the LiDAR to confirm the IP address remains configured.

Once this is done, the LiDAR should retain the IP even after powering off, so you won't need to switch back to Windows every time.

rostest, 2025-05-01:
Closed due to inactivity.

closed 2025-05-01 by rostest

Nervxz, 2025-05-10:
my lidar is MULS1AA-112211 multiScan136, icannot see store in device section

gschwsiSICKAG, 2025-05-12:
MULS1AA-112211 / multiScan136 is part of the multiScan100 family and is therefore supported:  
https://github.com/SICKAG/sick_scan_xd?tab=readme-ov-file#supported-sick-devices