SICKAG/sick_scan_xd #477: Is there some way to disable SICK reboot?

jardeldyonisio, 2025-04-15:
Hello,

We are using a SICK LSM100 sensor, and every time we run the launch file, the sensor reboots. This reboot process takes quite a long time, which is not ideal for our application.

After checking the code, it seems that a reboot is only necessary in case of certain errors. However, we would like to test the system without triggering a reboot each time.

rostest, 2025-04-15:
Your lidar probably starts in ASCII format and then switches to binary mode by restarting. You can preset the protocol with SOPAS ET and store it as the default value in the lidar. The lidar will then start immediately without having to change the protocol type first.

rostest, 2025-05-01:
Any updates on this? Otherwise we would like to close the ticket.

rostest, 2025-05-06:
closed due to inactivity.

rostest, 2025-05-19:
As @jardeldyonisio mentioned, he also observed this behaviour after setting the SICK LMS100 to CoLa binary communication.
We will investigate this with the following setup:

* Setting the SICK LMS100 via SOPAS ET to Binary CoLa communication (and storing this setting into EEPROM).
* Launch the LMS100 using the default launch file provided in the repository.
* We will use the following branch: https://github.com/SICKAG/sick_scan_xd/tree/feature/issue_update_2412xx.

jardeldyonisio, 2025-05-19:
> As [@jardeldyonisio](https://github.com/jardeldyonisio) mentioned, he also observed this behaviour after setting the SICK LMS100 to CoLa binary communication. We will investigate this with the [...]

Perfect!

rostest, 2025-06-12:
We have checked the start process on the LMS1xx again. You will find our start log attached. As you can see, there is no reboot. We therefore still think that you should switch the LMS1xx to binary mode in SOPAS ET. **Please do not forget to save the configuration data into the lidar after the changeover so that the change is persistent even after a cold start.**

Attachment (start log)

```
roslaunch sick_scan_xd sick_lms_1xx.launch hostname:=192.168.0.111
... logging to /home/rostest/.ros/log/7defd350-4758-11f0-a8f9-1fe6420a64da/roslaunch-ROS-NB-34941.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ROS-NB:42517/

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.15.8
 * /sick_lms_1xx/activate_lferec: True
 * /sick_lms_1xx/activate_lidinputstate: True
 * /sick_lms_1xx/activate_lidoutputstate: True
 * /sick_lms_1xx/add_transform_check_dynamic_updates: False
 * /sick_lms_1xx/add_transform_xyz_rpy: 0,0,0,0,0,0
 * /sick_lms_1xx/ang_res: 0.5
 * /sick_lms_1xx/cloud_topic: cloud
 * /sick_lms_1xx/encoder_mode: -1
 * /sick_lms_1xx/frame_id: cloud
 * /sick_lms_1xx/hostname: 192.168.0.111
 * /sick_lms_1xx/intensity: False
 * /sick_lms_1xx/intensity_resolution_16bit: False
 * /sick_lms_1xx/laserscan_topic: scan
 * /sick_lms_1xx/max_ang: 2.35619
 * /sick_lms_1xx/message_monitoring_enabled: True
 * /sick_lms_1xx/min_ang: -2.35619
 * /sick_lms_1xx/min_intensity: 0.0
 * /sick_lms_1xx/port: 2112
 * /sick_lms_1xx/range_filter_handling: 0
 * /sick_lms_1xx/range_max: 25.0
 * /sick_lms_1xx/range_min: 0.05
 * /sick_lms_1xx/read_timeout_millisec_default: 5000
 * /sick_lms_1xx/read_timeout_millisec_kill_node: 150000
 * /sick_lms_1xx/read_timeout_millisec_startup: 120000
 * /sick_lms_1xx/ros_qos: -1
 * /sick_lms_1xx/scan_freq: 50.0
 * /sick_lms_1xx/scandatacfg_timingflag: -1
 * /sick_lms_1xx/scanner_type: sick_lms_1xx
 * /sick_lms_1xx/start_services: True
 * /sick_lms_1xx/sw_pll_only_publish: True
 * /sick_lms_1xx/tf_base_frame_id: map
 * /sick_lms_1xx/tf_base_lidar_xyz_rpy: 0,0,0,0,0,0
 * /sick_lms_1xx/tf_publish_rate: 10.0
 * /sick_lms_1xx/tick_to_timestamp_mode: 0
 * /sick_lms_1xx/timelimit: 5
 * /sick_lms_1xx/use_binary_protocol: True
 * /sick_lms_1xx/use_generation_timestamp: True
 * /sick_lms_1xx/user_level: 3
 * /sick_lms_1xx/user_level_password: F4724744

NODES
  /
    sick_lms_1xx (sick_scan_xd/sick_generic_caller)

auto-starting new master
process[master]: started with pid [34951]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 7defd350-4758-11f0-a8f9-1fe6420a64da
process[rosout-1]: started with pid [34962]
started core service [/rosout]
[... 262 lines cut ...]
[ INFO] [1749710620.161594965]: Receiving: <STX>sRA SCdevicestate \x03<ETX>
[ INFO] [1749710620.161717289]: Waiting for scanner ready state since 33 secs
[ INFO] [1749710621.162130829]: Sending  : <STX><STX><STX><STX><Len=0017>sRN SCdevicestate CRC:<0x30>
[ INFO] [1749710621.164024516]: Receiving: <STX>sRA SCdevicestate \x01<ETX>
[ INFO] [1749710621.164190502]: Scanner ready for measurement after 34 [sec]
[ INFO] [1749710621.164319234]: Sending  : <STX><STX><STX><STX><Len=0017>sEN LMDscandata 0x01 CRC:<0x33>
[ INFO] [1749710621.165901251]: Receiving: <STX>sEA LMDscandata \x01<ETX>
[ INFO] [1749710621.185348135]: SickScanServices: service "/sick_lms_1xx/ColaMsg" created ("/sick_lms_1xx/ColaMsg")
[ INFO] [1749710621.186891592]: SickScanServices: service "/sick_lms_1xx/ECRChangeArr" created ("/sick_lms_1xx/ECRChangeArr")
[ INFO] [1749710621.188296766]: SickScanServices: service "/sick_lms_1xx/LIDoutputstate" created ("/sick_lms_1xx/LIDoutputstate")
[ INFO] [1749710621.189687881]: SickScanServices: service "/sick_lms_1xx/SCdevicestate" created ("/sick_lms_1xx/SCdevicestate")
[ INFO] [1749710621.191319134]: SickScanServices: service "/sick_lms_1xx/SCreboot" created ("/sick_lms_1xx/SCreboot")
[ INFO] [1749710621.193070183]: SickScanServices: service "/sick_lms_1xx/SCsoftreset" created ("/sick_lms_1xx/SCsoftreset")
[ INFO] [1749710621.194751719]: SickScanServices: service "/sick_lms_1xx/SickScanExit" created ("/sick_lms_1xx/SickScanExit")
[ INFO] [1749710621.194834961]: SickScanServices: ros services initialized
[ INFO] [1749710621.194938741]: Setup completed, sick_scan_xd is up and running. Pointcloud is published on topic "cloud"
[ INFO] [1749710621.197731538]: Software PLL locking started, mapping ticks to system time.
[ INFO] [1749710621.197813508]: 1 / 6 packets dropped. Software PLL not yet locked.
[ INFO] [1749710621.198344885]: 2 / 6 packets dropped. Software PLL not yet locked.
[ INFO] [1749710621.216110633]: 3 / 6 packets dropped. Software PLL not yet locked.
[ INFO] [1749710621.236195762]: 4 / 6 packets dropped. Software PLL not yet locked.
[ INFO] [1749710621.256479819]: 5 / 6 packets dropped. Software PLL not yet locked.
[ INFO] [1749710621.276641252]: Software PLL is ready and locked now!

```

jardeldyonisio, 2025-06-12:
> I think that maybe @jardeldyonisio have not expressed clearly. The "long time" is not necessarily a reboot process, he means the "waiting for scanner ready state since" warn. In our application it go up all the way to a minute and a half sometimes.

No, the issue is with the reboot process. Every time we launch the sensor, it reboots. We're looking for a way to prevent this behavior. @rostest We're using ROS2 here, is there any difference we should consider?

rostest, 2025-06-23:
As we have already noted, we are relatively certain that you have not permanently switched the lidar to COLA binary.

If there is still something open here, please send us a start log, as we have done (see ticket history).
There are no special characteristics with regard to ROS2.

rostest, 2025-07-24:
The behavior shown in your log is absolutely fine.

From line 225 onwards, the lidar electronics check whether the speed etc. in the lidar is correct. This check is repeated until the lidar reports OK. This usually takes a few seconds. In your case, 21 seconds, which is not unusual for this lidar (typically: 10 to 30 seconds).

In line 308 the start communication is successfully completed.

We will close the ticket again at the beginning of next week. We have opened it again for transparency reasons.

closed 2025-07-29 by rostest