SICKAG/sick_scan_xd #478: the pointCloud angle(Tim240) is different with differetn binary setting in driver 3.6.0

nancy-zhang-hub, 2025-04-18:
[image]

[image]

[image]

[image]

we update the sick driver form 2.0 to 3.6.0
then found out that, the binary setting affect the angle of pointCloud (cloud) which from sick driver. 
why?

now we add the transform angle in our launch file for tim240, and set binary false(well default false in driver 2.0, and default true in the new driver), to be compatible with all the previous devices that use 240.

[image]

and if the binary setting is different between launch file and tim240, will be more and more ‘good bye’ form sick driver (sick_generic_caller)

Jonny0816, 2025-04-30:
@rostest  This ticket had been submited 2 weeks but without any update, please help to check it, Thanks.

rostest, 2025-05-01:
Hi,

Please use the latest branch: `feature/issue_update_2412xx`.

We have understood your issue as follows — could you please confirm?

1. **Lidar:** You are using a TiM240, correct?  
2. **Coordinate system rotation:** In version 2.0, the point cloud is rotated 90 degrees compared to version 3.6.  
3. **Transformation:** You’ve applied a 90-degree rotation as a workaround. Is that correct?  
4. **Binary protocol:** With `use_binary_protocol=false`, you experience more frequent disconnects with the message “Good Bye.” Is that right?  
5. **Branch testing:** Does the `feature/issue_update_2412xx` branch show the same behavior?

Once we have your feedback, we will coordinate with SICK management next week to determine how to proceed with the ticket.

Thanks in advance!

rostest, 2025-05-01:
See also https://github.com/SICKAG/sick_scan_xd/issues/408

nancy-zhang-hub, 2025-05-05:
1. Lidar

 yes, we use TIM240

2.Coordinate system rotation: In version 2.0, the point cloud is rotated 90 degrees compared to version 3.6.

yes, we use the same launch file for tim240, just change this one place: 0 for sick_driver 2.0, 1.570796 for sick_driver 3.6.0
3.Transformation, we need to apply 90 degree rotation with new sick_driver 3.6.0(no need with driver 2.0)

 

4.With use_binary_protocol=true, we experience more frequent disconnects with the message “good bye.”, the default factory setting for TIM 240 is not binary, different with your default setting in driver 3.60

   we need to keep same with the setting of TIM240, so we changed to  use_binary_protocol=false. then, need add 3, apply 90 degree rotation.

5. Branch testing: Does the feature/issue_update_2412xx branch show the same behavior?

 yes, same to driver v3.6.0

cmake_minimum_required v3.13 in feature/issue_update_2412xx branch

cmake_minimum_required v3.5 in driver v3.6.0

rostest, 2025-05-06:
Thank you for your reply. We will look into this issue in the next few days (due date: 15 May).

rostest, 2025-05-13:
We have investigated the matter in detail and can tell you the following:

**Prerequisite:**
* Branch: feature/issue_update_2412xx
* Lidar: Tim240
* ROS version: ROS1 (noetic)
* Operating system: Ubuntu 20.04

**Behaviour:**
* Default launch file: Pointcloud displays correctly with the default launch file. The communication protocol is binary.
This is the recommended setting. Launch file (default) entry:
```
        <param name="use_binary_protocol" type="bool" value="True"/>
````

[image]

* Modified launch file with ASCII protocol. 
If the entry `use_binary_protocol` is set to `False`, as per the following example, the point cloud is rotated by 90 degrees. This does not correspond to the intended behaviour:
```       
        <param name="use_binary_protocol" type="bool" value="False"/>
 ```       

[image]

* In addition, we have observed that automatic switching from ASCII-protocol to binary-protocol or from binary-protocol to ASCII-protocol cannot be carried out cleanly with the TiM240. Therefore, the protocol type that is preset in the firmware by SOPAS ET should be used. Alternatively, the driver must be started twice. The protocol type is then set correctly on the first call and the driver terminates. On the second call, the correct type is already set internally and the driver starts correctly.

**Actions/Hints:**
  * Short term:
     * As user set the protocol to binary
     * Takes the standard launch file for the TiM240 as a reference
     * Use the above mentioned feature branch

  * Medium term:
     * SICK: Fix the wrong angle rotation if the driver is started in ASCII mode. We will add this to our backlog.

  * Currently not addressed:
     Full support for automatic conversion from binary to ASCII or from ASCII to binary without driver termination will not be implemented for the TiM240.

nancy-zhang-hub, 2025-05-14:
wow, you've found the problem

question is: 

1.we have hundreds of robots equipmented with tim 240, and use the ASCII protocol in SOPAS ET( it's also the default setting of tim 240 form you)

2.well use binary protocol,  the driver must be started twice.  and if the robot restart(reboot, power off--power on), also need to be started twice, right? it's unacceptable!

why are you assist on the binary setting in your diver( different wiht your factory setting on tim 240 firmware), does the binary has any advantageous?

if so, do you have any wireless method to change the setting in firmware, to change it form ASCII to binary, use the cable connect to sensor and open the SOPAS on cumputer to change the setting is not a possible way for hundreds of tim240

we use the sick driver 3.6.0 now, for all tim240 and picoscan 150

we import new sensors(picoscan 150), new driver(form v2.0 to v3.6), and also responsible for hundreds robots in factory at the same time.

 your suggestions is right, but we can't implement

image.png (view on web)

image.png (view on web)

rostest, 2025-05-22:
After further investigation we found the problem in the file `sick_generic_parser.cpp`.

Problem is this code line: 
https://github.com/SICKAG/sick_scan_xd/blob/develop/driver/src/sick_generic_parser.cpp#L1419

The line calculate the start angle for the ASCII protocol:
`msg.angle_min = (float)((starting_angle / 10000.0) / 180.0 * M_PI - M_PI / 2);`

Normally you have to subtract pi/2, as the X-axis points to the right in the SICK lidar system. With the TiM240, however, the X-axis also points forwards in SICK logic. Therefore, pi/2 must not be subtracted here. 

For a quick fix, you can use the following code addition as an idea, for example:
```
    msg.angle_min = (float)((starting_angle / 10000.0) / 180.0 * M_PI - M_PI / 2);
    // quick hack to compensate pi/2 offset for TiM 240
    if (basicParams[scannerIdx].getScannerName() == SICK_SCANNER_TIM_240_NAME)
    {
      msg.angle_min += (float)(M_PI / 2);
    }
```

However, instead of these quick changes, we will equalize the calculation for the start angle for the ASCII protocol and the binary protocol. The calculation in the binary protocol was already correct.

nancy-zhang-hub, 2025-05-23:
Thanks for your research. you've finally identified this issue

well, i'll also refrain form using the temporary fix for now, and continue using ASCII + 90

I'm not entirely clear about your conclusion. Are you planning to:

1.Update the drive to resolve this problem?

2.Adjust the angle definition specifically for TIM240?(in sick dirver or in sops of tim240)

if it's a dirver update, we'll wait for your upgrade before considering a complete resolution on our end.

https://github.com/SICKAG/sick_scan_xd/blob/develop/driver/src/sick_generic_parser.cpp#L1419

rostest, 2025-05-24:
Thank you for your query, for which we would like to provide the following explanation:

* The device firmware of the TiM240 will not be changed. The use of the X-axis orientation will remain as it is from a device perspective.

* The driver software will be adapted and the changes will be published as a patch release in the next few weeks.

closed 2025-07-13 by rostest