SICKAG/sick_scan_xd #480: Q: I have a question. How to use Tim781S on STM32 MUC.

wanilly, 2025-04-29:
We use STM32 f407 series. And so, we acquire Tim781s of the point cloud. Based on the source code, it seems to be implemting a Telegram command with TCP communication, but can I use Tim781s if I do the same with STM32?

rostest, 2025-04-30:
STM32 is not directly supported. The easiest way to create a bar metal application for yourself is to reconstruct the TCP communication. We recommend the following:

* Create an executable application on a supported platform based on the generic driver or ROS1.
* Record the communication with Wireshark.
* Export the Wireshark communication according to the following screenshot:

[image]

* Open this export in a text editor.
* Analyse the communication and compare it with the console output.
* If you have any questions, read the [telegram specification](https://cdn.sick.com/media/docs/7/27/927/telegram_listing_telegram_listing_ranging_sensors_lms1xx_lms5xx_tim2xx_tim5xx_tim7xx_lms1000_mrs1000_mrs6000_nav310_ld_oem15xx_ld_lrs36xx_lms4000_lrs4000_multiscan100_picoscan100_en_im0045927.pdf) from SICK and/or analyse the source code.
  
Finally, the communication is broken down as follows:
* Start: client/server communication (client: your application, server: the lidar) to configure the lidar.
* Then switch to ‘ listen’ mode. Your application receives the data telegrams and analyses them.

rostest, 2025-05-01:
@wanilly 
Closed because it is a little off topic. If you have any further feedback on this, please feel free to get in touch.