SICKAG/sick_scan_xd #482: LRS 4581R angle output range reducing  NO ROS,  C++ dll

sisusch, 2025-05-06:
Hello everyone

I am interfacing with the sensor in question. But I would like to understand how to configure the .launch file to obtain the same result that I obtain when using the web interface.
Below is what I would like to obtain, in which the measurement range is from -90 to +90 °. I would like to reduce the data that each sensor transmits to reduce traffic in the network and above all not have to delete most of the data in my app.

[image]

Below is the .launch configuration file I am using, in particular this two parameter i need an explanation on how to achieve the angle restriction with this parameter 

        <param name="min_ang" type="double" value="-1,57079"/> 
        <param name="max_ang" type="double" value="1,57079"/>   

```
<?xml version="1.0"?>
<!-- Using node option required="true" will close roslaunch after node exits -->
<launch>
    <arg name="hostname" default="192.168.1.201"/>
    <arg name="cloud_topic" default="cloud"/>
    <arg name="laserscan_topic" default="scan"/>
    <arg name="frame_id" default="cloud"/>
    <arg name="scan_cfg_list_entry" default="1"/>
    <arg name="skip" default="0"/>
    <arg name="nodename" default="sick_lrs_4xxx"/>
    <arg name="add_transform_xyz_rpy" default="0,0,0,0,0,0"/>
    <arg name="add_transform_check_dynamic_updates" default="false"/> <!-- Note: dynamical updates of parameter add_transform_xyz_rpy can decrease the performance and is therefor deactivated by default -->
    <arg name="encoder_mode" default="-1"/> <!-- Encoder settings (default: -1, i.e. not set): 0: Off, 1: Single increment, 2: Direction recognition phase, 3: Direction recognition level, 4: Fixed increment speed/ticks (LMS4000 only) -->
    <arg name="tf_publish_rate" default="10.0" />                     <!-- Rate to publish TF messages in hz, use 0 to deactivate TF messages -->
    <node name="$(arg nodename)" pkg="sick_scan_xd" type="sick_generic_caller" respawn="false" output="screen" required="true">
        <param name="scanner_type" type="string" value="sick_lrs_4xxx"/>
        <param name="hostname" type="string" value="$(arg hostname)"/>
        <param name="cloud_topic" type="string" value="$(arg cloud_topic)"/>
        <param name="laserscan_topic" type="string" value="$(arg laserscan_topic)"/>
        <param name="frame_id" type="str" value="$(arg frame_id)"/>
        <param name="port" type="string" value="2112"/>
        <param name="timelimit" type="int" value="5"/>
        <!-- <param name="min_ang" type="double" value="-3.1415926"/>  -->
        <!--<param name="max_ang" type="double" value="3.1415926"/>   -->

        <param name="min_ang" type="double" value="-1,57079"/> 
        <param name="max_ang" type="double" value="1,57079"/>   

        <param name="use_binary_protocol" type="bool" value="true"/>
        
        <!-- Optional range filter configuration: If the range of a scan point is less than range_min or greater than range_max, the point can be filtered. -->
        <!-- Depending on parameter range_filter_handling, the following filter can be applied for points with a range not within [range_min, range_max],   -->
        <!-- see enumeration RangeFilterResultHandling in range_filter.h:                                           -->
        <!--   0: RANGE_FILTER_DEACTIVATED,  do not apply range filter (default)                                    -->
        <!--   1: RANGE_FILTER_DROP,         drop point, if range is not within [range_min, range_max]              -->
        <!--   2: RANGE_FILTER_TO_ZERO,      set range to 0, if range is not within [range_min, range_max]          -->
        <!--   3: RANGE_FILTER_TO_RANGE_MAX, set range to range_max, if range is not within [range_min, range_max]  -->
        <!--   4: RANGE_FILTER_TO_FLT_MAX,   set range to FLT_MAX, if range is not within [range_min, range_max]    -->
        <!--   5: RANGE_FILTER_TO_NAN        set range to NAN, if range is not within [range_min, range_max]        -->
        <!-- Note: Range filter applies only to Pointcloud messages, not to LaserScan messages.                     -->
        <!-- Using range_filter_handling 4 or 5 requires handling of FLT_MAX and NAN values in an application.      -->
        <param name="range_min" type="double" value="0.0"/>
        <param name="range_max" type="double" value="100.0"/>
        <param name="range_filter_handling" type="int" value="0"/>
        
        <param name="intensity" type="bool" value="False"/>
        <param name="min_intensity" type="double" value="0.0"/> <!-- Set range of LaserScan messages to infinity, if intensity < min_intensity (default: 0) -->
        <param name="encoder_mode" type="int" value="$(arg encoder_mode)"/> <!-- Encoder settings (default: -1, i.e. not set): 0: Off, 1: Single increment, 2: Direction recognition phase, 3: Direction recognition level, 4: Fixed increment speed/ticks (LMS4000 only) -->
        <param name="use_generation_timestamp" type="bool" value="true"/> <!-- Use the lidar generation timestamp (true, default) or send timestamp (false) for the software pll converted message timestamp -->
        <param name="filter_echos" type="int" value="2"/> <!-- FREchoFilter settings: 0 = first echo, 1 = all echos (echo filter off), 2 = last echo -->
        <param name="skip" type="int" value="$(arg skip)"/> <!-- Default: 0 (i.e. publish each scan), otherwise only each n.th scan is published -->
        <param name="scan_cfg_list_entry" type="int" value="$(arg scan_cfg_list_entry)"/>
        <!-- Parameter scan_cfg_list_entry can be set to one of the following modes of the LRS4000 table (sets the scan configuration by "sMN mCLsetscancfglist <mode>")        
        Mode Name Interlaced ScanFreq ResultScanFreq Resolution TotalResol FieldOfView        
         1 12.5Hz & 0.040 deg 0x 12.5 Hz 12.5 Hz 0.040 deg 0.040 deg 360 deg
         2 12.5Hz & 0.060 deg 0x 12.5 Hz 12.5 Hz 0.060 deg 0.060 deg 360 deg
         4 12.5Hz & 0.100 deg 0x 12.5 Hz 12.5 Hz 0.100 deg 0.100 deg 360 deg
[... 220 lines cut ...]
[Info]: Receiving: <STX>sRA LMPscancfg \x00\x00\x04\xe2\x00\x01\x00[Lidar 0]: \x00\x01\x90\xff\xe4\x88\xc0\x00\x1b\x77\x40<ETX...
[Info]: LMPscancfg reply: "sRA LMPscancfg \x00\x00\x04\xe2\x00\x01\x00\x00\x01\x90\xff\xe4\x88\xc0\x00\x1b\x77\x40"
[Info]: LMPscancfg: { scan_frequency=1250, active_sector_cnt=1, angular_resolution[0][Lidar 0]: =400, start_angle[0]=-1800000, stop_angle[0]=1800000 }
[Info]: Angle resolution of scanner is 0.04 [deg]  (in 1/10000th deg: 400)
[Info]: MIN_ANG (after command verification): -3.14159 [rad] -180 [deg]
[Info]: MAX_ANG (after command verification): 3.14[Lidar 0]: 159 [rad] 180 [deg]
[Info]: Sending  : <STX><STX><STX><STX><Len=0018>sRN LMPoutputRange CRC:<0x5e>
[Info]: Receiving: <STX>sRA LMPoutputRange \x00\x01\x00\x00\x01\x90\xff\xe4\x88\xc0\x00\x1b\x77\x40<ETX>
[Info]: Angle resolution of scanner is 0.04 [deg[Lidar 0]: ]  (in 1/10000th deg: 400)
[Info]: MIN_ANG (after command verification): -3.14159 [rad] -180 [deg]
[Info]: MAX_ANG (after command verification): 3.14159 [rad] 180 [deg]
[Info]: Sending  : <STX><STX><STX><STX><Len=0018>sRN LMDscandatacfg CRC:<0x67>
[In[Lidar 0]: fo]: Receiving: <STX>sRA LMDscandatacfg \x01\x00\x00\x00\x00\x01\x00\x00\x00\x00\x00\x00\x01<ETX>
[Info]: Sending  : <STX><STX><STX><STX><Len=0018>sWN FREchoFilter 0x02 CRC:<0x7d>
[Info]: Receiving: <STX>sWA FREchoFilter <ETX>
[Info]: Sending  : <STX><[Lidar 0]: STX><STX><STX><Len=0016>sMN LMCstartmeas CRC:<0x68>
[Info]: Receiving: <STX>sAN LMCstartmeas \x00<ETX>
[Info]: Sending  : <STX><STX><STX><STX><Len=0007>sMN Run CRC:<0x19>
[Info]: Receiving: <STX>sAN Run \x01<ETX>
[Info]: Sending  : <STX><STX><STX><STX[Lidar 0]: ><Len=0017>sEN LMDscandata 0x01 CRC:<0x33>
[Info]: Receiving: <STX>sEA LMDscandata \x01<ETX>
[Lidar 0]: LIDAR inizializzato correttamente con config: sick_lidar_0.launch
```

kind regards.

rostest, 2025-05-06:
The first step is to correct the typo in your launch file:

```
    <param name="min_ang" type="double" value="-1,57079"/> 
    <param name="max_ang" type="double" value="1,57079"/>   
``` 
should be:
```
    <param name="min_ang" type="double" value="-1.57079"/> 
    <param name="max_ang" type="double" value="1.57079"/>   
```

sisusch, 2025-05-07:
Hi Rostest,

your suggestion solves my problem, in the configuration file to separate the whole part I have to use the point. Copy and paste from the Italian calculator puts the comma.

Thanks

<param name="min_ang" type="double" value="-1.57079"/> 
<param name="max_ang" type="double" value="1.57079"/>

closed 2025-05-07 by sisusch