SICKAG/sick_scan_xd #484: Possible to run ROS2 driver in passive UDP-only mode?

mferenduros-ocado, 2025-05-21:
[This code](https://github.com/SICKAG/sick_scan_xd/blob/develop/driver/src/sick_generic_laser.cpp#L497) seems to suggest that the laser driver supports running in UDP-only mode, but I don't see how to actually trigger that code path - it seems like `rosGetParam(nhPriv, "hostname", hostname)` would always succeed because the `hostname` param has been declared with a default value.

Am I missing something? Is there a way to bring up the driver such that useTCP remains false and it does not attempt to make SOPAS connections?

FWIW we're talking to a picoscan120

gschwsiSICKAG, 2025-05-21:
Hi @mferenduros-ocado, could you please provide a bit more context on your use case? Why are you aiming to use a "UDP-only" mode?

Just for background: the **picoScan120** uses **TCP** for parameterization and **UDP** for streaming measurement data. This setup is fixed and is the **only supported combination** when using the **picoScan120 together with the sick_scan_xd driver**.

rostest, 2025-07-13:
Closed due to clarification.

closed 2025-07-13 by rostest