SICKAG/sick_scan_xd #485: ROS2 Jazzy support in lastet release v3.6.0

kinni318, 2025-05-23:
Dear 
I hope this message finds you well.
I would like to ask whether the latest release(3.6.0) already support ROS 2 Jazzy.
if not, could you please share any plans or timeline for Jazzy compatibility?

Thank you...
Kin

rostest, 2025-05-24:
Hi, thanks for your interest in sick_scan_xd and for raising this issue.

You're right — up to now, official testing has been limited to ROS 2 versions up to and including Humble. However, starting with release [v3.5.0](https://github.com/SICKAG/sick_scan_xd/releases/tag/3.5.0), we’ve added support for bloom releases targeting newer ROS 2 distributions, including Iron and Jazzy.

That said, while we’ve made necessary adjustments for compatibility, full regression testing on Jazzy has not yet been completed. We do expect the driver to work on Jazzy, assuming the underlying dependencies are satisfied (e.g., proper launch_ros, rclcpp, etc.). If you encounter any issues while running on Jazzy, please report them — contributions and test feedback are very welcome.

kinni318, 2025-05-30:
Hi, 
I sincerely appreciate your response. it is very helpful.

Best regards,

closed 2025-05-30 by kinni318