SICKAG/sick_scan_xd #486: HI!! I have a one question...

wanilly, 2025-05-28:
`2025-05-31T06:26:41.041830+00:00 ubuntu taskset[151146]: terminate called after throwing an instance of 'std::runtime_error'
2025-05-31T06:26:41.042205+00:00 ubuntu taskset[151146]:   what():  normalizeSecNSec: time (4294967295,4294967295) is out of dual 32-bit range`

We use the tim781S device. I find this error. Why occure this error...  

Is it software-wise wrong to go over the range of values, or is the mode wrong in the launch file? 

`<param name="scanner_type" type="string" value="sick_tim_7xxS"/>
        <param name="nodename" type="string" value="$(arg nodename)"/>
        <param name="min_ang" type="double" value="-2.35619449"/> 
        <param name="max_ang" type="double" value="2.35619449"/>  
        <param name="use_binary_protocol" type="bool" value="true"/>
        
        
        
        
        
        
        
        
        
        
        
        
        <param name="range_min" type="double" value="0.0"/>
        <param name="range_max" type="double" value="100.0"/>
        <param name="range_filter_handling" type="int" value="0"/>
        
        <param name="intensity" type="bool" value="True"/>
        <param name="hostname" type="string" value="$(arg hostname)"/>
        <param name="cloud_topic" type="string" value="$(arg cloud_topic)"/>
        <param name="laserscan_topic" type="string" value="$(arg laserscan_topic)"/>
        <param name="frame_id" type="str" value="$(arg frame_id)"/>
        <param name="port" type="string" value="$(arg port)"/>
        <param name="skip" type="int" value="$(arg skip)"/> 
        <param name="timelimit" type="int" value="5"/>
        <param name="sw_pll_only_publish" type="bool" value="$(arg sw_pll_only_publish)"/>
        <param name="use_generation_timestamp" type="bool" value="true"/> 
        <param name="start_services" type="bool" value="True"/> 
        <param name="activate_lferec" type="bool" value="True"/>         
        <param name="activate_lidoutputstate" type="bool" value="True"/> 
        <param name="activate_lidinputstate" type="bool" value="True"/>  
        <param name="active_field_set" type="int" value="-1"/> 
        <param name="field_set_selection_method" type="int" value="-1"/> 
        <param name="min_intensity" type="double" value="0.0"/> 
        <param name="scandatacfg_timingflag" type="int" value="-1"/> 

        
        
         
        
        
        
        <param name="add_transform_xyz_rpy" type="string" value="$(arg add_transform_xyz_rpy)" /> 
        <param name="add_transform_check_dynamic_updates" type="bool" value="$(arg add_transform_check_dynamic_updates)" />

        <param name="message_monitoring_enabled" type="bool" value="True" />      
        <param name="read_timeout_millisec_default" type="int" value="5000"/>     
        <param name="read_timeout_millisec_startup" type="int" value="120000"/>   
        <param name="read_timeout_millisec_kill_node" type="int" value="150000"/> 
        
        <param name="client_authorization_pw" type="string" value="6FD62C05"/>    

        
        
        
        
        
        
        <param name="ros_qos" type="int" value="-1"/>  

        
        <param name="tf_base_frame_id" type="string" value="map" />              
        <param name="tf_base_lidar_xyz_rpy" type="string" value="0,0,0,0,0,0" /> 
        <param name="tf_publish_rate" type="double" value="$(arg tf_publish_rate)" />                

        
        <param name="initialize_scanner" type="int" value="$(arg initialize_scanner)"/>
        
        
        <param name="tick_to_timestamp_mode" type="int" value="2"/>`

rostest, 2025-05-28:
I have renamed the issue for better internal allocation. Please use the branch 
```
https://github.com/SICKAG/sick_scan_xd/tree/feature/issue_update_2412xx
```
This branch should fix the problem.
Alternatively, you can set the following setting to 0:
```
tick_to_timestamp_mode = 0
```

Setting tick_to_timestamp_mode = 0 tells sick_scan_xd not to translate sensor ticks into precise timestamps, and instead rely on default system timing. This is simpler but may be less accurate for time-critical applications like sensor fusion.

See also ticket #440.

wanilly, 2025-05-29:
Hello! Thank you for your response. but, this branch delete...?? 

We not found this page..

https://github.com/SICKAG/sick_scan_xd/tree/feature/issue_update_2412xx

Umm... Is this branch is merged..???

rostest, 2025-05-29:
We merged this branch yesterday. Please try using the `develop` branch by default now: https://github.com/SICKAG/sick_scan_xd/tree/develop.

wanilly, 2025-06-02:
Thanks you for your response...

But, I use branch develop code... The builded lib is not execute lib... If you tell me the code for error resolution about the related code, I'll fix it.

Additionally, we have these issues.

[image]

wanilly, 2025-06-04:
Thank you for your response... I was mistaken.

closed 2025-06-05 by rostest