SICKAG/sick_scan_xd #491: Run lidar multiscan 100/136 error

Nervxz, 2025-06-09:
hello, i have run the lidar multiscan through command ros2 launch sick_scan_xd sick_multiscan.launch.py hostname:=<ip-address> udp_receiver_ip:=<ip-address> on linux ubuntu 22.04 jetson orin ros2  but it have error like below, when i plug into another machine running ros 2 but not jetson it still working, how can i fix this
[sick_generic_caller-1] [WARN] [1749464578.901834958] [sick_scan]: diagnostic_updater: No HW_ID was set. This is probably a bug. Please report it. For devices that do not have a HW_ID, set this value to 'none'. This warning only occurs once all diagnostics are OK. It is okay to wait until the device is open before calling setHardwareID.

rostest, 2025-06-10:
Can you give us an example (Log, screenshot) of how we can reproduce this message?

Nervxz, 2025-06-11:
when i plug back to my linux PC I can run it easy, however on jetson I cannot even though I have same directory of sick_scan_xd, I even pull the new one from SICK but cannot make it

rostest, 2025-06-12:
Even if Jenkins is not one of the supported platforms, we are pleased that you were able to start the driver successfully. The message you see is just a warning that you can ignore.

If it really bothers you, you can add the following code part to `sick_generic_caller.cpp`:
```
std::string current_hw_id = updater.getHardwareID();
if (current_hw_id.empty() || current_hw_id == "none" || current_hw_id == "unknown") {
 updater.setHardwareID("unknown"); // or simply "none" if nothing better is available
}
```
We are closing this ticket as it relates to unsupported hardware and think we have given you a hint on how to solve the problem. You are also welcome to submit a pull request if you were able to resolve the warning as described above.