SICKAG/sick_scan_xd #493: Is field evaluation monitoring for Piscoscan150 planned ?

fahdonj, 2025-06-23:
Hi,
As mentioned in the README (click [here](https://github.com/SICKAG/sick_scan_xd?tab=readme-ov-file#field-evaluation-information)), only LMS1xx, LMS5xx, TiM7xx and TiM7xxS support field evaluation monitoring in the ros driver. 
[image]

I'm working on a project that uses a Picoscan150, which doesn't support this feature yet.

Is field evaluation monitoring for Piscoscan150 planned ? If not, would it be possible to add it ?

Thanks!

fahdonj, 2025-07-01:
Hi @gschwsiSICKAG , thanks for your reply! I’ll look into your suggestion regarding the use of ROS services.

rostest, 2025-07-13:
Closed due to clarification.

closed 2025-07-13 by rostest