SICKAG/sick_scan_xd #499: picoscan120 publishes unintended frame_id

cjh1995-ros, 2025-07-22:
Hi, I'm sick developer! Currently I'm developing robot with the two picoscan120. Everything is fine but I have one problem that `frame_id` of picoscan120 is unintended one. I set `publish_frame_id` as `lidar_front`. But the real frame id of it is `lidar_front_1`. So I checked the ROS1 launch file and found this sentence.

```xml
    <arg name="publish_frame_id" default="world" />                                 <!-- frame id of ros Laserscan messages, default: "world_<layer-id>" = "world_1" -->

...

        <!-- Use "1" to activate resp. "0" to deactivate laserscan messages (picoScan has just 1 layer) -->
        <param name="laserscan_layer_filter" type="string" value="1" />
```

So I think the `frame_id` of picoscan120 must be `frame_id`_1. But I want to change it from `frame_id`_1 to `frame_id`. How can I do it?

rostest, 2025-07-22:
Hi, thanks for your question — this is closely related to [issue #500](https://github.com/SICKAG/sick_scan_xd/issues/500).

As a short-term workaround, you can disable the automatic frame suffixing behavior by modifying the following line in the code:

🔗 [ros_msgpack_publisher.cpp:880](https://github.com/SICKAG/sick_scan_xd/blob/develop/driver/src/sick_scansegment_xd/ros_msgpack_publisher.cpp#L880)

After making the change, be sure to:

- Clean all intermediate build and development directories (`build`, `devel`, etc.)
- Rebuild your workspace completely

cjh1995-ros, 2025-07-29:
Thank you for your answer! We will check ASAP and tell you the result!

rostest, 2025-08-19:
Closed due to inactivity.

closed 2025-08-19 by rostest