SICKAG/sick_scan_xd #502: ROS2 build broken and small fix for ROS2 Kilted needed

Darkproduct, 2025-07-24:
The package is currently broken for ROS2 because of 07f032aadb6bca9236c87fecda69ed250efcbb68 which introduced some ROS1 message namespaces. 

https://github.com/SICKAG/sick_scan_xd/blob/07f032aadb6bca9236c87fecda69ed250efcbb68/include/sick_scan/sick_lmd_scandata_parser.h#L80-L84

Additionally, `tf2` switched from `*.h` headers to `*.hpp` in Kilted, which breaks here:
```diff
--- a/include/sick_scan/sick_ros_wrapper.h
+++ b/include/sick_scan/sick_ros_wrapper.h
@@ -266,7 +266,7 @@ public:
 #include <std_msgs/msg/color_rgba.hpp>
 #include <visualization_msgs/msg/marker.hpp>
 #include <visualization_msgs/msg/marker_array.hpp>
-#include <tf2/LinearMath/Quaternion.h>
+#include <tf2/LinearMath/Quaternion.hpp>
 #include <tf2_ros/transform_broadcaster.h>

 typedef rclcpp::Node::SharedPtr rosNodePtr;
```

**To build it on kilted** I applied the above diff and `git revert 07f032aadb6bca9236c87fecda69ed250efcbb68`

rostest, 2025-07-26:
Thank you very much for your valuable feedback regarding the Kilted version. 
We will gladly review it, reproduce it if necessary, and implement a solution, as “Kilted” is already on our release to-do list.

Darkproduct, 2025-07-31:
Looking good. But I think this fix is not only needed for Kilted but for all ROS2 versions, because in all of them the message namespace has changed.

Regarding the fix, at least on the current rolling version, the above mentioned change to the tf2 include header is needed, see this error log:

> In file included from /home/dark/ws/kamaro/beteigeuze_ws_rolling/src/sick_scan_xd/include/sick_scan/abstract_parser.h:39,
>                  from /home/dark/ws/kamaro/beteigeuze_ws_rolling/src/sick_scan_xd/driver/src/abstract_parser.cpp:63:
> /home/dark/ws/kamaro/beteigeuze_ws_rolling/src/sick_scan_xd/include/sick_scan/sick_ros_wrapper.h:269:10: fatal error: tf2/LinearMath/Quaternion.h: No such file or directory
>   269 | #include <tf2/LinearMath/Quaternion.h>
>       |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~

We currently have to build the `sick_scan_xd` package by hand, because there is no binary release on the rolling upstream for this package. So it would be great, if we would be able to continue that without having to maintain our own fork.

closed 2025-09-17 by rostest