SICKAG/sick_scan_xd #506: Abnormal angle_max Published by Picoscan150

Blezy725, 2025-07-29:
Thank you for your support in issue #500. 
I have found a new issue and would like to ask for your advice.

In my program, I need to correct `angle_max` and `angle_min` in the `LaserScan` data for rotation purposes, 
so these values need to be reasonably accurate.

While monitoring the `LaserScan` values using `rostopic echo`, 
I noticed that `angle_max` have abnormal values.

According to the launch file, I expect `angle_min` and `angle_max` to be ±138.0 [deg] (±2.4086... [rad]), 
but the published `angle_max` value appears to be incorrect.

Is there any way to avoid or correct this behavior? 
Any guidance would be greatly appreciated.

my config
```
[ INFO] [1753788242.811430600]: sick_generic_caller V. 3.7.0 githash:"06c4f748e774367ba2726611316f63a9d3f5c938" gitinfo:"2025-05-28 10:57:12 +0200      Alexander Weinmann      Merge branch 'release/3.7.0'"  
[ INFO] [1753788242.816753100]: Program argument 1: /home/ubuntu/sick_lidar_ws/devel_isolated/sick_scan_xd/lib/sick_scan_xd/sick_generic_caller
[ INFO] [1753788242.817094700]: Program argument 2: __log:=/home/ubuntu/.ros/log/857680f2-6c6e-11f0-9d82-84a93896edf1/sick_picoscan-2.log
[ INFO] [1753788242.817412500]: Program argument 3: ~scan_fullframe:=scan
[ INFO] [1753788242.817685500]: Program argument 4: ~scan_fullframe:=scan
[ INFO] [1753788242.817947900]: ==========================================
[ INFO] [1753788242.859305900]: SickTransformPublisher: broadcasting TF messages parent_frame_id="map", child_frame_id="base_link", (x,y,z,roll,pitch,yaw)=(0.0,0.0,0.0,0.0,0.0,0.0), rate=10.0
[ INFO] [1753788242.965633200]: SickCloudTransform: add_transform_xyz_rpy = (0,0,0,0,0,0)
[ INFO] [1753788242.965916200]: SickCloudTransform: azimuth_offset = 0 [deg]
[ INFO] [1753788242.966106000]: SickCloudTransform: additional 3x3 rotation matrix = { (1,0,0), (0,1,0), (0,0,1) }
[ INFO] [1753788242.966270800]: SickCloudTransform: apply 3x3 rotation = false
[ INFO] [1753788242.966494200]: SickCloudTransform: additional translation = (0,0,0)
[ INFO] [1753788242.966679700]: SickCloudTransform: check_dynamic_updates = false
[ INFO] [1753788242.968565900]: sick_scansegment_xd configuration:
[ INFO] [1753788242.968828500]: scanner_type:                     sick_picoscan
[ INFO] [1753788242.969039700]: udp_sender:
[ INFO] [1753788242.969258300]: udp_port:                         2115
[ INFO] [1753788242.969456300]: check_udp_receiver_ip:            1
[ INFO] [1753788242.969644600]: check_udp_receiver_port:          2116
[ INFO] [1753788242.969830800]: all_segments_min_deg:             -138
[ INFO] [1753788242.970006800]: all_segments_max_deg:             138
[ INFO] [1753788242.970182000]: publish_frame_id:                 base_link
[ INFO] [1753788242.970376000]: publish_laserscan_segment_topic:  scan_segment
[ INFO] [1753788242.970523600]: publish_laserscan_fullframe_topic:scan_fullframe
[ INFO] [1753788242.970663900]: udp_input_fifolength:             20
[ INFO] [1753788242.970803200]: msgpack_output_fifolength:        20
[ INFO] [1753788242.970937900]: verbose_level:                    0
[ INFO] [1753788242.971115700]: measure_timing:                   1
[ INFO] [1753788242.971261000]: export_csv:                       0
[ INFO] [1753788242.971404300]: export_udp_msg:                   0
[ INFO] [1753788242.971538300]: logfolder:
[ INFO] [1753788242.971663300]: hostname:                         192.168.0.10
[ INFO] [1753788242.971786600]: udp_receiver_ip:                  192.168.0.50
[ INFO] [1753788242.971927500]: udp_timeout_ms:                   10000
[ INFO] [1753788242.972079500]: udp_timeout_ms_initial:           60000
[ INFO] [1753788242.972214900]: scandataformat:                   2
[ INFO] [1753788242.972336700]: performanceprofilenumber:         -1
[ INFO] [1753788242.972484000]: imu_enable:                       0
[ INFO] [1753788242.972621300]: imu_topic:                        imu
[ INFO] [1753788242.972779200]: imu_udp_port:                     7503
[ INFO] [1753788242.972955700]: imu_latency_microsec:             0
[ INFO] [1753788242.973126600]: sopas_tcp_port:                   2111
[ INFO] [1753788242.973334600]: start_sopas_service:              1
[ INFO] [1753788242.973483100]: send_sopas_start_stop_cmd:        1
[ INFO] [1753788242.973635800]: sopas_cola_binary:                0
[ INFO] [1753788242.973772800]: sopas_timeout_ms:                 5000
[ INFO] [1753788242.973901100]: host_read_filtersettings:         1
[ INFO] [1753788242.974029000]: host_FREchoFilter:                0
[ INFO] [1753788242.974199100]: host_set_FREchoFilter:            1
[ INFO] [1753788242.974334100]: host_LFPangleRangeFilter:         0 -138.0 +138.0 -90.0 +90.0 1
[ INFO] [1753788242.974462100]: host_set_LFPangleRangeFilter:     0
[ INFO] [1753788242.974587700]: host_LFPlayerFilter:
[ INFO] [1753788242.974739500]: host_set_LFPlayerFilter:          0
[ INFO] [1753788242.974955900]: host_LFPintervalFilter:           0 1
[ INFO] [1753788242.975132300]: host_set_LFPintervalFilter:       0
[ INFO] [1753788242.975302500]: laserscan_layer_filter:           1
[ INFO] [1753788242.975445500]: msgpack_validator_enabled:                         0
[ INFO] [1753788242.975590000]: msgpack_validator_verbose:                         1
[ INFO] [1753788242.975859900]: msgpack_validator_discard_msgpacks_out_of_bounds:  1
[ INFO] [1753788242.976041300]: msgpack_validator_check_missing_scandata_interval: 9
[ INFO] [1753788242.976258600]: msgpack_validator_required_echos:                  0 1 2
[ INFO] [1753788242.976501700]: msgpack_validator_azimuth_start:                   -2.40855 [rad]
[ INFO] [1753788242.976747500]: msgpack_validator_azimuth_end:                     2.40855 [rad]
[ INFO] [1753788242.976935400]: msgpack_validator_elevation_start:                 0 [rad]
[ INFO] [1753788242.977139100]: msgpack_validator_elevation_end:                   0 [rad]
[ INFO] [1753788242.977312400]: msgpack_validator_valid_segments:                  0 1 2 3 4 5 6 7 8
[ INFO] [1753788242.977472800]: msgpack_validator_layer_filter:                    1

```

[image]

rostest, 2025-07-29:
Thank you for sharing this information with us. The “angle_increment” value corresponds to 1/20 deg, which also indicates an incorrect calculation. We will investigate this issue.

Blezy725, 2025-07-29:
Thank you for your reply.

As additional information, I am using the Picoscan150 Pro, 
running in the 0.05 [deg], 15 Hz mode.

Please let me know if you need any further details for your investigation.