SICKAG/sick_scan_xd #508: First connection to Tim-571 LiDAR fails with sick_scan_xd v3.7.0, second attempt succeeds

kjoelovelife, 2025-08-05:
Hello SICK team,

I am using two Tim-571 LiDARs with sick_scan_xd version 3.7.0. When powering on the LiDARs and launching the nodes, the first connection attempt always fails (the driver cannot connect to the devices). However, if I restart the nodes or relaunch after the first failure, the connection succeeds and the LiDARs work as expected.

Steps to reproduce:

Power on both Tim-571 LiDARs.
Launch the ROS 2 nodes using sick_scan_xd-3.7.0 (see launch file snippet below).
Observe that the first connection attempt fails.
Relaunch the nodes (without power cycling the LiDARs).
The connection now succeeds.
Expected behavior:
The LiDARs should connect successfully on the first attempt after power-on.

Actual behavior:
First connection attempt always fails; second attempt (without power cycling) succeeds.

Configuration:

Device: Tim571-2050101 (two units)
sick_scan_xd version: 3.7.0
ROS 2 distribution: Humble
OS: Ubuntu 22.04
Launch file: lidar_test.launch.py
Configuration files: first_lidar.yaml, second_lidar.yaml(same as the first_lidar.yaml but the node name is `second_scan_node`)

[lidar_test.zip](https://github.com/user-attachments/files/21594089/lidar_test.zip)

Zeug555, 2025-08-05:
Same issue here with the same LiDAR and ROS2 distro ! Can you notice me if you find a solution ?

rostest, 2025-08-07:
Thank you very much for your kind reply and detailed statement, which we will gladly refer to in future tickets of this kind. Much appreciated!