SICKAG/sick_scan_xd #509: Workaround for nav2_costmap_2d implementation detail

mferenduros-ocado, 2025-08-11:
sick_scan_xd sets `range_max` to the largest range reading in the given scan, and so it varies according to what the lidar can currently see. However nav2_costmap_2d has a feature that implicitly assumes range_max is a large, fixed value.

The feature deals with the problem that stale obstacles in the costmap can only be cleared by new rays passing through them (which positively detect that the obstacle is gone). But if there is no valid surface behind a stale obstacle, then the lidar will never report a valid ray passing through it and the obstacle will remain forever. So as a workaround, nav2_costmap_2d also supports using rays with range = +inf to clear obstacles.

This is done internally by replacing any +inf readings with `range_max-epsilon`. It is assumed that `range_max` will be a fixed value greater than the costmap's configured radius, and therefore the rays will only clear obstacles but not create new ones.

However if range_max ever falls below the costmap's radius, as can happen with sick_scan_xd, this assumption breaks and any +inf rays will generate invalid obstacles in the costmap.

Even though the driver is reporting perfectly valid data, and there is an easy workaround of making a custom filter to overwrite `range_max`, it would nonetheless be nice if the driver supported reporting a fixed value for `range_max`.

rostest, 2025-08-13:
@mferenduros-ocado

We have reviewed your detailed explanation and the source code you cited. We understand your request to mean that the value `range_max` can be set optionally in the launch file. If the entry is present there, then the current standard procedure is not selected. Instead, the values stored in the launch file are used for `range_max`.

Is this assumption correct? 

If ‘yes’, we will add it to our backlog.

mferenduros-ocado, 2025-08-16:
> Is this assumption correct? 
> 
> If ‘yes’, we will add it to our backlog.  

Yes that's exactly it, thank you