SICKAG/sick_scan_xd #512: Jitter of header timestamp

cmuehlbacher, 2025-08-22:
The timestamp in the message header seems to have a high jitter although the laser is spinning constantly. 

The setup is the following causing this issue: 
- ROS Humble 
- Sick Picoscan 150Pro FW version 1.5.1 
- NTP synchronization of the laser with the host PC 
- Scan frequence is 40Hz (but this problem also occurs with other frequencies) 

With the current mast version, the following time difference can be observed for the header timestamp. See the following image:
[image] 

Statistics about this: 
min 0.000665 
max: 0.050320 
avg: 0.025001 

The average is fine, but it jitters a lot. 

Without the PLL algorithm (basically skipping this adjustment) the time difference is much more consistent. See the following image:

[image] 

Statistics about this: 
min 0.023946 
max: 0.026866 
avg: 0.025002 

Therefor it seems not to be a firmware issue but an issue of the PLL algorithm, as bypassing this step yields much better results. 

We observed this behavior also with older driver version and on multiple laser scans (basically every scanner we checked). 

If needed, we can provide recordings to narrow down the problem.

rostest, 2025-08-22:
Hello,

Please let us know which version of the driver you are using. Please also check whether this behaviour also occurs with the current ‘develop’ version. 

The software PLL was introduced for older lidars to reduce jitter on the lidar side. If you do not experience any problems with picoScan time stamping after disabling the software PLL, you are welcome to disable the software PLL and run the driver in this mode.

All the best and good luck!

weinmalSICKAG, 2025-08-29:
I will close this issue for now, since disabling the software PLL seems to improve the timestamp jitter for you. Feel free to repoen the issue if it still an issue.

closed 2025-08-29 by weinmalSICKAG