SICKAG/sick_scan_xd #514: urdf missing for lidar multiScan 1xx

abegni, 2025-09-15:
Hi,
I'm using MULS1AA-112211 multiScan136. In folder urdf, any .xacro model about multiScan 1xx is not present. In meshes folder, .stl model about this type of lidar is also missing. Could it be possible to add at least the .xacro for multiScan 1xx?

Best regards,
Andrea

closed 2025-09-16 by rostest

rostest, 2025-09-17:
We understand your point. Would it be possible for you to create a xacro model based on the CAD model and make it available here in the ticket? We would greatly appreciate it.
We could then add it to the model list. If it is not suitable, we would also appreciate your feedback, as we would then need to clarify the priorities for this internally.

abegni, 2025-09-18:
Good idea but i'm workring on dimensional drawing of the lidar found it in the data sheet. I will define the exactly position of the axis orgin. I will provide the .xacro

[image]